Bobsilvio/epcube #1: Valori Potenzq

xpeppe86x, 2025-04-23:
Ciao,

tutti i valori potenza sono errati di un decimale (o per intenderci sono 1/10 del reale).
Ti allego lo screen con accanto le reali da app (tutto x10), poi onestamente non capisco perchè molti sensori diano la stessa cifra, ma facciamo test.

Peppe

[image]

Bobsilvio, 2025-04-23:
Grazie, nel confrontarli con i miei..escono esatti i miei.
Mi manderesti la foto della schermata che hai da epcube? https://monitoring-eu.epcube.com/

xpeppe86x, 2025-04-23:
> Bobsilvio

Eccoti gli screen presi uno dopo l’altro, come vedi è tutto X10 il real, non so perché mi prende i valori così…..anche in Watt non cambia è solo un moltiplicatore quello

￼￼

Bobsilvio, 2025-04-23:
non devi rispondere alla mail, scrivi direttamente da github, non vedo gli screen.

xpeppe86x, 2025-04-23:
Si scusa...... non avevo capito che la mail era autogenerata da qui.....

[image]

[image]

Bobsilvio, 2025-04-23:
controlla se funziona:
https://github.com/Bobsilvio/epcube/releases/tag/v1.0.4

xpeppe86x, 2025-04-23:
Ciao,
Mi sembrano di nuovo a vecchio tutte le traduzioni (nell'ultima avevi differenziato tutti i nomi tra potenza ed energia). Ma il problema rimane, continuo a vedere tutto 1/10

xpeppe86x, 2025-04-23:
[image]

[image]

Bobsilvio, 2025-04-23:
dopo che hai flaggato hai riavviato?

xpeppe86x, 2025-04-23:
Ciao,

Ho il flag, adesso il x1000 funziona (però era X10 da farevisto che sbagliava di un decimale non di 3).
Il problema è che lo fa su tutti i valori. Quindi sballa le Energie che invece erano corrette.
Andrebbe fatto x10 solo sui sensori delle potenze non quelli delle Energie che invece erano corretti
[image]
[image]

Bobsilvio, 2025-04-25:
Testa la 1.0.6, dovrebbe essere ok, i valori sono in Watt di Power..

xpeppe86x, 2025-04-25:
Ciao,
Ho provato , ho riavviato proprio  il sistema per scrupolo ma niente, tutti i valori di potenza sia che li imposti in W o KW sono sempre 1/10 del reale. Dovrebbero essere tutti i valori di potenza x10

xpeppe86x, 2025-04-25:
[image]

Bobsilvio, 2025-04-25:
scusa quanto stai consumando ora? se non metti la foto di paragone epcube app non capisco che valori vedi.

xpeppe86x, 2025-04-25:
Ciao, come ti dicevo tutto quello che leggi, la realtà è x10.
Dove segna 52w erano 520w, dove segnava 0.2xxx, la realtà era 2.xxx.
Tutto x10

[image]

xpeppe86x, 2025-04-27:
Ciao!
Non avevo visto il push della 1.7, ho visto che da codice hai aggiunto un *10 ed infatti adesso i valori della potenza li vedo corretti!

[image]
[image]

closed 2025-04-29 by Bobsilvio