Bobsilvio/epcube #2: Icona Batteria Homepage

xpeppe86x, 2025-04-23:
Ciao , ho notato anceh questo 👍 

[image]

Bobsilvio, 2025-04-23:
Si è un problema che ho notato, avendo vari valori batteria in giro prende quelli della configurazione del soc.
Da risolvere appena capisco come fare..

Bobsilvio, 2025-04-23:
fixato probabilmente:
https://github.com/Bobsilvio/epcube/releases/tag/v1.0.5

xpeppe86x, 2025-04-23:
Ciao,

Ho rimesso l'ultima ed ho riavviato, per scrupolo ho pure restartsto il sistema ma stessa cosa. Le traduzioni sono sempre anche quelle vecchie.[image]
[image]

Bobsilvio, 2025-04-23:
le traduzioni sono nuove.
Ma hai attivato l'opzione per il +1000 nella configurazione?

xpeppe86x, 2025-04-23:
Non so te, però adesso la batteria la vedo ok ! Fixed?

closed 2025-04-24 by Bobsilvio