Bobsilvio/epcube #3: Ottenere valore restituzione alla rete

alex3025, 2025-04-28:
Non mi sembra che sia presente un sensore che rappresenti i kWh ritornati alla rete.

Il valore nella dashboard proprietaria è questo:
[image]

Sono io ceco oppure non è ancora stato implementato?

alex3025, 2025-04-29:
Scusami tanto, non volevo apparire scontroso, ho aperto l'issue in velocità preso dalla foga di implementare questa integrazione e non sono stato molto attento.

Comunque, non mi sembra di vedere nessun sensore che riguardi il valore dell'immissione :/

Se ti serve qualche screenshot delle varie entità fammi sapere!

Scusa ancora e grazie tante per star sviluppando questa integrazione!

Bobsilvio, 2025-04-29:
Come gia detto i valori immissione sono:

'Rete Energia' --> totale immesso in rete oggi
'Rete Potenza' --> immissione live in rete

Almeno nel mio caso corrispondono.

alex3025, 2025-04-29:
Il mio valore di "Rete Energia" corrisponde a questo:

[image]

Bobsilvio, 2025-04-29:
ok continuo a non capire cosa cerchi allora. se mi posti poi solo il valore dal sito e non mi posti anche il valore dell'integrazione..come facciamo a compararlo?

alex3025, 2025-04-29:
Scusa, pensavo non fosse necessario dato che erano uguali, comunque ecco qua il valore del sensore:

[image]

Come puoi vedere, è (quasi, si è aggiornato ovviamente) uguale a quello nella foto che ho postato prima (dove sotto viene indicato che si riferisce a "Da Rete"), mentre io sto cercando il sensore che riporta il valore "Alla Rete" (quello nel primo messaggio).

Bobsilvio, 2025-04-30:
Allora, effettivamente mancava il sensor di prelievo dalla rete.. non restituzioni, sospetto che tu abbia la pinza girata al contrario nel quadro, mi sembra strano che prelevi 4.37 e immetti 0.27.. 
A prescindere da questo ora ci sono tutti i sensori:

https://github.com/Bobsilvio/epcube/releases/tag/v1.0.8

alex3025, 2025-04-30:
Quindi "Rete Energia" è il valore di immissione (da inserire nella sezione restituzione) mentre il valore di consumo dalla rete mancava, ho capito bene?

Questo è il mio stato attuale comunque:

[image]

[image]

Bobsilvio, 2025-04-30:
[image]

con l'aggiornamento vedrai questo. Mi sembra chiara la dicitura.
Mentre la cosa strana(che è offtopic) e che tu prelevi dalla rete 3.69 e immetti 0.01.. hai forse un impianto piccolino, o la pinza girata al contrario nel quadro elettrico che misura i valori al contrario..
cioè 3.69 è l'energia immessa e 0.01 è la prelevata.

Spero di essermi spiegato..

[image]

alex3025, 2025-04-30:
Si scusa per l'off-topic, l'epcube è stato aggiunto ad un impianto esistente (produzione ac), tra un paio di giorni dovrebbero arrivarmi anche i pannelli da collegare direttamente all'epcube (produzione dc).

La capacità totale una volta completati i lavori dovrebbe essere di circa 674W.

Bobsilvio, 2025-05-01:
issue risolto

closed 2025-05-01 by Bobsilvio