Bobsilvio/epcube #6: Avvisi con sensori batteria v1.1.0

alex3025, 2025-05-05:
Ciao! Dopo l'aggiornamento alla versione 1.1.0 e dopo aver ricreato l'integrazione (come scritto nelle release notes), andando ad aggiungere di nuovo la batteria nella dashboard energetica, sono apparsi i seguenti avvisi. 

[image]

Bobsilvio, 2025-05-07:
Grazie per la segnalazione, usa i valori cumulativi e non quelli giornalieri della v1.1.1
https://github.com/Bobsilvio/epcube/releases/tag/v1.1.1

alex3025, 2025-05-07:
Non ho capito se devo utilizzare la funzione `utility_meter` oppure sono presenti dei sensori già pronti nella nuova versione. Grazie in anticipo!

Bobsilvio, 2025-05-07:
sono gia pronti.

alex3025, 2025-05-07:
Ok, ho provato e lasciando i sensori come nella foto che ho mandato non ci sono più gli errori. È corretto?

Bobsilvio, 2025-05-07:
corretto

closed 2025-05-07 by Bobsilvio